rnd-ash/ultimate-nag52-fw #1: Document testing approach

rusefillc, 2021-12-25:
How is this firmware tested?

any unit tests of logic layer?
any info on any sorts of simulated hardware?

rnd-ash, 2021-12-26:
Currently nothing is simulated. However I will be working eventually on a desk testing setup! 

At this point in time my testing simply involves driving on private roads using very unfavorable conditions when I want to test something. Otherwise the TCM itself learns in most situation and can report back to me after I drive

rusefillc, 2022-01-17:
I was mostly curious about https://github.com/rnd-ash/ultimate-nag52-fw/tree/main/test which mentions unit tests. From my experience some automated testing is very helpful for complex systems and unit tests are amazing first layer of defense prior to any real hardware?