rnd-ash/ultimate-nag52-fw #5: Handle TCM reset whilst in motion

rnd-ash, 2022-02-27:


ilian-stoyanov, 2026-09-30:
For months I've been trying to contact any living person regarding this project. I wanted to ask if this is ready for prime time, as I have to say the effort put in marketing this product is completely inversely proportional to the one spent developing it.

I guess this is my answer. This is just a pasttime for the author. Engine-destroying issue not addressed for 4.5 years...

rnd-ash, 2026-09-30:
It's very much ready for prime time - Look at the work on the dev branch, with one caviat, you have to do some tweaking to the base maps yourself to get it to behave how you want, since the firmware can't be perfect for every car at the same time.

Admittedly I'm the author, but I have been happily using this since the early days, and now have about 100 users happily using it in their various vehicles.

I'll update the GitHub issues, those are definitely stale for ages

rnd-ash, 2026-10-03:
Closing issue. This is fixed. If the TCU is turned on when in motion, it engages hydraulic Neutral forcefully, blocks engine starting until gearbox enters P/N.

The rest of the work for this (To engage drive after a reset in motion) is in #4

closed 2026-10-03 by rnd-ash