rnd-ash/ultimate-nag52-fw #8: Race / launch control

rnd-ash, 2022-02-27:
The 722.9 automatic gearbox in the E63 AMG has a launch control feature. I think it is possible to copy this!

The user should only be able to do this in 'M' (Manual) mode!

1. Request engine Max RPM / Torque to limit engine RPM somewhere desirable

2. Whilst the engine is ramping up, pre-apply pressure to the torque converter clutch so we are biting but not 100% applied, this acts as a trans brake effectively, allowing the engines turbo/supercharger to build boost as the engine will be under a lot of load in this situation

3. When the user releases the brake pedal, quickly fully engage the torque converter clutch, creating a direct drive.

ShaneDelmore, 2022-07-15:
I did something similar with my old 1989 GMC Syclone with a 700R4 4 speed transmission but found that quickly engaging the torque limiter wasn't always the best option.  Even though the engine made around 700hp at the crank for a 3300lb vehicle, I got my best times at the drag strip not engaging the torque converter clutch until around 3000-3400rpm because engaging it any earlier meant I lost out on the torque multiplication of the converter and ended up accelerating more slowly.  The overall idea feels sound to me though, allow building boost from a stop, limit torque to prevent spinning the tires too quickly, then ramp up power as speed increases and more power can be handled without spinning the tires.

dzid26, 2022-08-15:
> 2. Whilst the engine is ramping up, pre-apply pressure to the torque converter clutch so we are biting but not 100% applied, this acts as a trans brake effectively, allowing the engines turbo/supercharger to build boost as the engine will be under a lot of load in this situation

Trans-brake typically means that you tie-up the gearbox by applying four clutches. Then you release one clutch to form first or second gear (depending on traction). 
Applying the torque converter clutch wouldn't make much sense at the beginning. If rpm rise to high for good start (depending on engine power and torque converter itself) you would limit the torque before launching. 
Applying torque converter clutch, as @ShaneDelmore said can be done at high rpm. For non-tuned engines that would be just before the redline. This will recoup a slight amount of engine inertia and help the shift. To help the shift happen slightly quicker you would open tcc again just after the ratio starts to change. That will of course also give some torque multiplication in the next gear.

rnd-ash, 2026-10-03:
Not possible without the use of a trans-brake (Which the 722.6 does not have). Closing

closed 2026-10-03 by rnd-ash