rnd-ash/ultimate-nag52-fw #23: M113 + M113K engine ignores torque request

rnd-ash, 2022-05-18:
Not sure if this is a bug with the way EGS52_CAN_HAL requests torque,
or if it is intended design for the AMG engines

The M113 and M113k engines (Found in 55 badged AMG vehicles) seem to ignore the TCMs torque requesting

We should have a config option that allows the TCM to know if the engine will ignore the request, and shift differently as a result in order to maintain shift smoothness

ShaneDelmore, 2022-07-15:
I wonder if that is only for some vehicles.  My 2011 G55 appears to have torque limiting working correctly, to the point it is annoying and in low range with a lot of traction I can't even generate enough power to turn the wheels and climb a ledge (while off-roading). I'll try to think of a good way to test this with my truck.

kestreltom, 2022-09-19:
I am almost certain there is a tune on your E55..  The tune is probably causing certain torque requests to be ignored.

rnd-ash, 2023-03-05:
This has been resolved as of d7d49e3d2e672584fba1c881ba4a07cdf311c2a2

It seems the whole issue was ME ECUs not ack.ing torque requests when the gearbox is in a fail state (CRD Diesel ECUs don't care). The TCU was not setting gearbox OK on boot!

closed 2023-03-05 by rnd-ash