rnd-ash/ultimate-nag52-fw #31: EGS CAN Matrix database for pre 2006 modules

rnd-ash, 2022-08-23:
Just to note all the different EGS CAN Matrix's from EGS/EGS51/EGS52 labelled part numbers

|MB PN|EGS Label type|Manufacture date|GS418|GS218|GS338|
| --- | --- | --- | --- | --- | --- |
|A0215451632|EGS51|99W33|2 bytes|6 bytes|N/A|

viper-666, 2023-01-09:
On my Chrysler Crossfire it looks like that:
<html>
<body>

MB PN | EGS type | Manf. date | Simens code | EWM/TRRS | GS218 | GS418 | GS338
-- | -- | -- | -- | -- | -- | -- | --
A 033 545 13 32 | EGS52 | 06W08 | 5WP2000JRB | EWM | 8 bytes | 8 bytes | 8 bytes

</body>
</html>
But GS338 look a litte strange:  338#FFFF1FFF00FF0000

rnd-ash, 2023-01-15:
Awesome, thanks for that data! Yes, GS338 does look weird until the car starts moving. It seems like there are some fields in GS338 that is not turbine speed, but has no description in the CAN DBC (So is likely not used by any ECU in the car)

viper-666, 2023-01-17:
> Awesome, thanks for that data! Yes, GS338 does look weird until the car starts moving. It seems like there are some fields in GS338 that is not turbine speed, but has no description in the CAN DBC (So is likely not used by any ECU in the car)

Ok, as I read the GS338 the engine wasn't running.