rnd-ash/ultimate-nag52-fw #50: Future Request: Use W/S switch for changing of the driving profile

viper-666, 2023-01-28:
On the switching  lever of these cars: Chrysler Crossfire, Mercedes SLK R170, Mercedes W208 + W202 C, 2007 SSANGYOUNG RODUIS is near the switching lever a switch where you can choose between the Winter and Summer driving profile. As these cars don't have a switch for other driving profiles like sport, agility ect. it isn't possible to change the driving profile of the Ultimate NAG52.  The only possibility is with the config_app, but that's not realy a solution.
So I ask if you can integrate this switch, it will be transferred on CAN Address 230 the first bit is for the w/s switch 0 = S and 1 = W the other 7 bits are for the position of the switcher so D = 000010101 in summer and D = 100010101 when the winter switch is on.
But change only the driving profile when the bit goes from 0 -> 1  elsewhere if the user let in in the w position the driving profile will change all time.

rnd-ash, 2023-01-30:
So, based on the CAN data
```
ECU EWM
	FRAME EWM_230h (0x00000230)
		SIGNAL W_S, OFFSET: 0, LEN: 1, DESC: Driving program, DATA TYPE BOOL
		SIGNAL FPT, OFFSET: 1, LEN: 1, DESC: Driving program button actuated, DATA TYPE BOOL
		SIGNAL KD, OFFSET: 2, LEN: 1, DESC: Kickdown, DATA TYPE BOOL
		SIGNAL SPERR, OFFSET: 3, LEN: 1, DESC: barrier magnet energized, DATA TYPE BOOL
		SIGNAL WHC, OFFSET: 4, LEN: 4, DESC: gear selector lever position (NAG only), DATA TYPE ENUM
			ENUM D, RAW: 5, DESC: selector lever in position "D"
			ENUM N, RAW: 6, DESC: selector lever in position "N"
			ENUM R, RAW: 7, DESC: selector lever in position "R"
			ENUM P, RAW: 8, DESC: selector lever in position "P"
			ENUM PLUS, RAW: 9, DESC: selector lever in position "+"
			ENUM MINUS, RAW: 10, DESC: selector lever in position "-"
			ENUM N_ZW_D, RAW: 11, DESC: selector lever in intermediate position "N-D"
			ENUM R_ZW_N, RAW: 12, DESC: selector lever in intermediate position "R-N"
			ENUM P_ZW_R, RAW: 13, DESC: selector lever in intermediate position "P-R"
			ENUM SNV, RAW: 15, DESC: selector lever position unplausible
```

It looks like the bit is `W_S`. Where 1 is W and 0 is S mode.

Now, I need to think of implementation method. Since it is not a button, and UN52 has more than 2 profiles. Do I make it so that you can configure 2 profiles (1 for W, 1 for S mode)?, or do I make it so that every time you toggle the switch the profile goes to the next profile?

viper-666, 2023-01-30:
Please make it so, that it toggles between the profiles.
Thank you.

closed 2023-02-17 by viper-666