rnd-ash/ultimate-nag52-fw #74: FEATURE - MAXX / Other custom ECU byte aligned CAN layer

rnd-ash, 2024-01-03:
The issue? EGS CAN Layers use bit stuffing, and thus, certain values are not byte aligned.

Certain custom ECUs do not support bit stuffing on CAN, thus, it is not feasible to port any of the EGS CAN layers to custom ECUs.

This issue will be used to discuss requires Tx/Rx communication and build up a CAN spec file for custom ECUs.

rnd-ash, 2024-01-03:
To work on this issue, I'll need to know what CAN IDs do these ECUs transmit data on, and what kind of data can they receive.

Also, in such a configuration, what shifter is used, and how is rear wheel speed obtained? (Physical sensor or by the engine ECU)

Magnup, 2024-01-03:
I'll answer the best I CAN;)

Maxxecu have a default CAN protocol found here: https://www.maxxecu.com/webhelp/can-default_maxxecu_protocol.html

But it also supports 36 custom CAN "input values" and 10 custom outgoing messages. 

Outgoing messages can be up to 8 bytes, big or small endian, signed or unsigned, 8, 16 or 32 bit. These can send basically any 
value within the ECU with an offset, multiplier and divider value. 
There is also a possibility to create 3 custom bytes where the bits are set to 1 or 0 always, or by any live data within the ECU. These bytes can be sent as part of any of the custom outgoing messages.  

"Input values" means, as far as I can understand, Maxxecu can extract 36 values from 36 or less messages. values can be signed or unsigned, 8, 16 or 32 bit, big or small endian,. It is possible to use a mask for example to use only 14 bit of a 16 bit message. Data recieved can also be given an offset, multiplier and divider value. 

Maxxecu can receive wheel speed either by some OEM CAN protocols (not mercedes) or by physical sensors at the wheels or driveshaft (as I have).

I have the old PRND4321 (TRRS?) shifter but will probably change to the PRND +/- shifter and I am guessing that would be the one most would use.

rnd-ash, 2024-04-09:
Initial draft of this is merged into dev firmware. 

The protocol is documented here. We can add data as we see fit at later dates

https://docs.ultimate-nag52.net/en/advanced/custom-can

closed 2026-10-03 by rnd-ash